roboticslab-uc3m/kinematics-dynamics #148: [ICartesianControl.h] Think about more than manipulator arms

jgvictores, 2018-03-02:
[ICartesianControl.h] Long-term proposal, to merge `movj` and `movl` resulting in something like:
```cpp
gotoPose(CartesianInterpolation::MoveJ,xd);
```

More generic Cartesian targets even for drones!

PeterBowman, 2023-02-15:
On the interface layer, choosing between two input enums as in `gotoPose(CartesianInterpolation::MoveJ, xd)` and `gotoPose(CartesianInterpolation::MoveL, xd)` would be pretty much equivalent to supporting two methods as we currently do: `movj(xd)`, `movl(xd)`.

Now on the implementation layer: how about drones, i.e. free-moving 3D solids? You don't need a kinematic solver here, but rather a cartesian trajectory generator. For linear trajectories, KDL is the best fit. But I'm not really convinced of supporting both kinematic chains (more generally speaking: articulated bodies) and simple solid objects in BasicCartesianControl. I'd rather abstract out the logic necessary for the latter, i.e. implement my own trajectory generator. The interface design of [`ICartesianControl`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/488028bcb8d9043c3cd6f208de4eded5e857e4ef/libraries/YarpPlugins/ICartesianControl.h) simply doesn't fit for a drone, I think.

@jgvictores can we close this?

jgvictores, 2023-02-16:
Yes, I completely agree. I've been doing some more work with mobile robots over the past years, and concerns are typically very different. Abstraction makes no sense in this case, as the objectives and terminology differ quite a bit (localization, navigation...).

Closing!

closed 2023-02-16 by jgvictores