roboticslab-uc3m/kinematics-dynamics #155: Promote compile-time USE_LMA option to runtime usage in KdlSolver

PeterBowman, 2018-06-12:
As spoken with @jgvictores, it would be nice to convert the `USE_LMA` CMake option (handled as a preprocessor definition, [ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/da0a4ac2aefe94a607418adbabc9108e8251d70c/libraries/YarpPlugins/KdlSolver/ICartesianSolverImpl.cpp#L11-L16)) into a YARP RF option loadable along with the device itself (via command line or programatically).

Not sure about the capitalization: is `--useLMA` ok? And sticking to current defaults (this option is [set to `OFF`](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/da0a4ac2aefe94a607418adbabc9108e8251d70c/libraries/YarpPlugins/KdlSolver/CMakeLists.txt#L22)), such property wouldn't accept any value and the **lack** of a `--useLMA` shall mean *use the Newton-Raphson algorithm*.

jgvictores, 2018-06-12:
I was thinking more of an `--ik lma`, leaving the future open for more than 2 options.

jgvictores, 2018-06-12:
I'd almost default to `lma`. : )

BTW, I recall a fork with a LMA_JL too, but haven't been able to find it. However, here's something that I have found: [KDL::ChainIkSolverPos_LMA_JL_Mimic Class](http://docs.ros.org/lunar/api/moveit_kinematics/html/classKDL_1_1ChainIkSolverPos__LMA__JL__Mimic.html), [chainiksolver_pos_lma_jl_mimic.h](http://docs.ros.org/lunar/api/moveit_kinematics/html/chainiksolver__pos__lma__jl__mimic_8h_source.html).

jgvictores, 2018-06-12:
PS: Just re-discovered https://github.com/orocos/orocos_kinematics_dynamics/pull/35 (Added joint limits to ChainIkSolverPos_LMA): apparently closed but never merged...

jgvictores, 2018-06-12:
Regarding ` KDL::ChainIkSolverPos_LMA_JL_Mimic`, so much [WET development](https://github.com/search?p=2&q=ChainIkSolverPos_LMA_JL_mimic&type=Code) would deserve some effort for a PR on KDL (unless I'm really missing something).

PeterBowman, 2018-06-22:
Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/8c82f75c63549cc217d4b5dcd3db8d9afb713728.

closed 2018-06-22 by PeterBowman

jgvictores, 2018-07-12:
Note to self: check out what is done in https://github.com/ros-industrial-consortium/descartes (has refs to both KDL and ikfast).

jgvictores, 2018-07-12:
Also: https://rosindustrial.org/news/2018/7/5/optimization-motion-planning-with-tesseract-and-trajopt-for-industrial-applications