roboticslab-uc3m/kinematics-dynamics #156: Implement a dry run mode in 'movj' and 'movl'

PeterBowman, 2018-06-28:
From https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/135#issuecomment-359866382:

> Idea: perform a dry run on the desired trajectory (usually `movl`) and check for any limit violations, e.g. maximum joint velocity might be exceeded at some point or the operation would take too long (the latter does not apply to current strategies since trajectory duration is one of our input parameters). On success, perform the trajectory; on failure, return immediately and notify callers.

jgvictores, 2019-03-12:
This idea is nice. It's like an "offline" checker. It would be great if the JMC could do this (kind of like `movj` vs `inv`), but our JMC are unfortunately unaware of trajectories. I understand this is what it pushes it out of the JMC, and, to not end up with another layer in between, goes to the catch-all BCC.
1. From what I'm seeing, maybe `BasicCartesianControl` should be renamed `BasicTrajectoryContol` or something.
2. Can't completely discard the joint-space offline checker idea... feels kind of like a collision checker that works with the trajectory before it's launched....

PeterBowman, 2019-05-01:
Regarding offline cartesian/joint trajectories:

* https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/124#issuecomment-488339423
* https://github.com/roboticslab-uc3m/yarp-devices/issues/208

PeterBowman, 2022-06-13:
Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/26f8c35f4253cc984bc326cbb8ee9fe4b515f4ae. To enable this, launch BCC with `enableFailFast`. Internally, IK is performed first, then joint position limits are checked (somewhat innecessarily since the IK should fail earlier), then joint velocity limits.

closed 2022-06-13 by PeterBowman