roboticslab-uc3m/kinematics-dynamics #162: Review joint sizes shared between controller and solver

PeterBowman, 2018-07-19:
BasicCartesianControl integrates two distinct members related to the number of controlled joints:

https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/BasicCartesianControl/BasicCartesianControl.hpp#L229

* `numRobotJoints` is the number of joints as regarded by the joint controller (e.g. CanBusControlboard). On TEO, this is 7 (6 actuated rotary joints + 1 "fake" gripper joint).
* `numSolverJoints` is the number of joints as regarded by the cartesian solver (e.g. KdlSolver). On TEO, this is 6 (the number of non-fixed joints **after** the kinematic chain is configured).

In the KDL world, the latter is returned by `KDL::Chain::getNrOfJoints()`. However, it's also important to note that certain solvers work with the number of **segments** (links), as returned by `KDL::Chain::getNrOfSegments()`.

Since most data travels between API calls inside a container of some sort (`std::vector`), and its size is not enforced in any way, hard-to-spot errors may arise.

* At the BasicCartesianControl level, a 7-element vector of external forces is instantiated and initialized here:

  https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/BasicCartesianControl/RateThreadImpl.cpp#L244-L252

  The first six force vectors (applied on each link) are null/empty, just the seventh one matters and it should be interpreted as a force exerted on the tool's TCP. Note that there are only six joints in the "standard" chain (gripper aside). KDL sees six joints, but eight links (kinematic chain prepended with H0, and HN appended thereafter). Inverse dynamics are performed on said vector ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/BasicCartesianControl/RateThreadImpl.cpp#L256)). However, the KDL implementation of ID requires to fill a vector of *wrenches*, built from the initial vector of external forces (esentially, their meaning is the same):

  https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/KdlSolver/ICartesianSolverImpl.cpp#L297-L305

  The weird part here is that `idsolver.CartToJnt(...)` ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/KdlSolver/ICartesianSolverImpl.cpp#L311)) expects a vector of eight external forces, that is, `wrenches.size() == 8`. As explained earlier, `wrenches[0]` should correspond to a force exerted on H0 (fixed joint), and `wrenches[7]` (in the case described here) to a force exerted on HN (another fixed joint). It becomes apparent that the assignment to `wrenches[i]` inside the loop ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/KdlSolver/ICartesianSolverImpl.cpp#L301)) should read `wrenches[i+1]` instead, size checks aside.

  Note: check how exactly KDL treats external forces. Is `fexts[6]` applied on the tip of the tool's TCP (probably), or perhaps on the COG of the last link?

* Compare:

  https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/BasicCartesianControl/RateThreadImpl.cpp#L254

  ...and:

  https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c187abdc0c9b2284d36df0f22cf64ecb238642b2/libraries/YarpPlugins/KdlSolver/ICartesianSolverImpl.cpp#L323

  Assuming that `numRobotJoints` equals to `7` (see previous point), `t[6]` now probably points to an unhandled (unassigned, perhaps used by another variable) memory block and trying to access it might result in undefined behavior. In fact, `t.at(6)` will throw an exception, while `t[6]` always spits out whatever resides at that location in memory.

  There are several places throughout the code where this is relevant.

PeterBowman, 2019-09-17:
I plan to merge `numRobotJoints` and `numSolverJoints` into one entity that represents motor joints, only. That is, the (until and for now) only additional YARP-handled joint, which corresponds to the LacqueryFetch single-motor gripper, will be exposed via a dedicated YARP port to which BasicCartesianControl would have to connect via a separate remote_controlboard device in case it needs to access actuator commands (https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/115). See rationale at https://github.com/roboticslab-uc3m/yarp-devices/issues/228#issuecomment-532200863.

jgvictores, 2019-09-18:
Two comments related to the iCub:
1) Specifically on the iCub, I recall the hand joints are coupled to the arm. I know we do not have one, but maybe it is not a great idea to make our controller incompatible with its port layout.
2) In YARP's http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#a425a689b2ab34bf2d13f1a17197a2ccc class, there is a differentiation between controlled joints and blocked joints (joints that are not controlled). Maybe this kind of mapping would be better?

PeterBowman, 2019-09-18:
Ad 1:

Did we ever comply with iCub regarding joint or cartesian control? Are we compatible with iCub in the current state of this repo? If so, what should we keep in mind in order to keep being compliant, e.g. are there any written guidelines for that respect?

Ad 2:

We handle blocked (non-actuated) joints in the solver layer, every axis passing through the BCC is actuated (including gripper joints). In our case, the first N joints belong to a kinematic chain, whereas gripper joints always live at the end of this joint list. Doesn't seem the case for the method you link to. If you really want to speak about randomly inserted blocked joints, in the current state of things, I think this should be discussed at the joint controller level (yarp-devices repo).

PeterBowman, 2019-09-18:
> If you really want to speak about randomly inserted blocked joints, in the current state of things, I think this should be discussed at the joint controller level (yarp-devices repo).

Just recalled https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/178.

PeterBowman, 2022-06-15:
> I plan to merge `numRobotJoints` and `numSolverJoints` into one entity that represents motor joints, only.

Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/83b02a2ab551912fb9bf23eb155f500ef6fefa29.

PeterBowman, 2022-06-16:
> * The first six force vectors (applied on each link) are null/empty, just the seventh one matters and it should be interpreted as a force exerted on the tool's TCP. Note that there are only six [...]

Considering this, @jgvictores, would you mind if we 🔥 **BREAK THE API** 🔥 (just a little bit)? I'd suggest to replace

```cxx
ICartesianSolver::invDyn(const std::vector<double> &q, const std::vector<double> &qdot, const std::vector<double> &qdotdot,
                         const std::vector<std::vector<double>> &fexts, std::vector<double> &t);
```

with

```cxx
ICartesianSolver::invDyn(const std::vector<double> &q, const std::vector<double> &qdot, const std::vector<double> &qdotdot,
                         const std::vector<double> &ftip, std::vector<double> &t);
```

That is, expect `const std::vector<double> &ftip` instead of `const std::vector<std::vector<double>> &fexts`. While `fexts` is a vector of external wrenches applied on each segment/link, `ftip` would refer to its last element: external forces applied on the tip.

Why? Because we can't query the number of chain/tree segments as it is required by the current API. Currently, this is hardcoded as I explained in the first comment, and maybe even wrong or buggy. In case we load a kinematics model with more or less segments, the loop in PeriodicThreadImpl.cpp could segfault. Therefore:

* either we add `ICartesianSolver::getNumSegments()` (or `getNumLinks`) and use this information to instantiate `fexts`, or
* we modify the `ICartesianSolver` API and let it fill the vector of wrenches with zeros conveniently.

Both alternatives assume that we won't ever want to provide non-zero values for external forces applied on other segments (besides the tip). Moreover, the signature of `ICartesianControl::forc` don't allow to do otherwise (the input vector is `ftip`).

jgvictores, 2022-06-17:
Let's go for it!

PeterBowman, 2023-01-28:
Important finding: the diagram represented [in the wiki](https://www.orocos.org/wiki/Kinematic_Trees.html) does not depict the actual behavior of KDL solvers accurately enough.

[image]

The Fcog is, in fact, referred to the tip frame of the segment. In other words, the "reference frame" stated [in the documentation of `RigidBodyInertia`](https://github.com/orocos/orocos_kinematics_dynamics/blob/5541147d4a220cab97d0ae1efa1aa860557d5c32/orocos_kdl/src/rigidbodyinertia.hpp#L40-L44) is the tip frame, not the local base/root frame.

Further reading: https://github.com/orocos/orocos_kinematics_dynamics/issues/170 and https://www.orocos.org/forum/orocos/orocos-users/problem-dynamic-model-and-inverse-dynamic-solver-kdl.

PeterBowman, 2023-01-29:
Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/be7289c125da5160815563a29a6076ab448b1cc1.

closed 2023-01-29 by PeterBowman

PeterBowman, 2023-01-30:
The interpretation of external/exerted forces is a source of confusion.

* `int KDL::ChainIdSolver::CartToJnt(const KDL::JntArray &q, const KDL::JntArray &q_dot, const KDL::JntArray &q_dotdot, const KDL::Wrenches &f_ext, KDL::JntArray &torques)`: here, `f_ext` is a vector of external wrenches **applied on** each segment
* https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/f9523fc080ac9e70a497e8ea35946a23cf8cd257/libraries/YarpPlugins/ICartesianSolver.h#L202-L203 here, `f_tip` is a single wrench **applied on** the TCP
* https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/f9523fc080ac9e70a497e8ea35946a23cf8cd257/libraries/YarpPlugins/ICartesianControl.h#L263 here, `fd` is a single wrench **exerted by** the TCP

The last one requires that sign inversion is performed when passing data between controller and solver. This is done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/f9523fc080ac9e70a497e8ea35946a23cf8cd257.