roboticslab-uc3m/kinematics-dynamics #168: Create a generator-enabled trajectory

jgvictores, 2018-12-13:
Create a [generator](https://en.wikipedia.org/wiki/Generator_(computer_programming))-enabled trajectory.
 
I've become very fond of the [generator](https://en.wikipedia.org/wiki/Generator_(computer_programming)) design pattern, and I think it makes sense with 2 scenarios we are going to find:
- Trajectories generated offline but can be "theoretically infinite", such as following a straight line, or pre-programmed pendulum movements or walking gaits.
-Trajectories generated online. 

Extra points if the generator is wrapped and can be programmed outside of C++, aka Python!

PS: This idea is in line with #135, oriented at designing and running more complex trajectories, so blocked by #134 as well.

jgvictores, 2018-12-13:
By the way, still not sure if a new class, or reusing an old class and setting duration to a reserved negative number, etc.

PeterBowman, 2021-03-15:
> Offline generated trajectories that can be "theoretically infinite", such as following a straight line, or pre-programmed pendulum movements or walking gaits.

Following a straight line (at constant speed) is implemented in `movv`: https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/170.

> Online generated trajectories.

And this one should be achieved with plain `movi`.

> By the way, still not sure if a new class, or reusing an old class and setting duration to a reserved negative number, etc.

Idea: don't over-complicate BasicCartesianControl and prefer external trajectory generators. I would wrap my generator routine in a `yarp::os::PeriodicThread`/`yarp::os::Timer` and make it speak to the cartesian controller via CartesianControlClient.

jgvictores, 2021-03-17:
When I opened this issue, it was an idea inspired by the generator pattern I saw in Python, as I believed there could be some easy and portable similar mechanism in C++. It turns out that SO points out some (imho) pretty convoluted implementations ([ref](https://stackoverflow.com/questions/9059187/equivalent-c-to-python-generator-pattern)).

Maybe some day C++ will support Python-style generators with a proper `yield` keyword and a `next` member. This assumption may not be so far-fetched, e.g. ES6 Javascript adopted a similar mechanism (<https://codeburst.io/understanding-generators-in-es6-javascript-with-examples-6728834016d5>), but I'm not sure we are anywhere near that day in the C++ development roadmap.

TLDR: How about closing this issue as a `wontfix`, and instead focus on the mini-"infrastructure" with examples of working code such as the `yarp::os::PeriodicThread`/`yarp::os::Timer` mechanism you mention?

PeterBowman, 2021-03-17:
I agree that iterators are (and have always been) the C++-ish way to go, at least for now. I am not sure though about the scope of this issue. Where and how is this hypothetical generator supposed to fit? I'm not convinced to have it embedded in BasicCartesianControl since it would probably require another rework on the ICartesianControl interface. As I see it, such more advanced trajectory generation techniques are better placed outside the controller (separation of concerns), be it C++ or Python code. In that case, you can trick KDL motion API into generating endless trajectories: https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/134#issuecomment-794102173. More complex stuff involves resorting to `for` loops, as usual.

In other words, I think you can already have "generation-enabled" trajectories today.

> TLDR: How about closing this issue as a `wontfix`

Since you agree, I'm proceeding to do so. Please feel free to re-open and/or provide further details on your proposal.

PS see [exampleScrewTheoryTrajectory](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/63871757390c48ab6f33b77a902e09a0d914e25c/examples/cpp/exampleScrewTheoryTrajectory) regarding the implementation of a basic `PeriodicThread` mechanism.

closed 2022-05-22 by PeterBowman