roboticslab-uc3m/kinematics-dynamics #175: Stream acquisition timestamp along with FK data in BCC/CCS

PeterBowman, 2019-04-16:
Our cartesian controller device (server) streams the result of `ICartesianControl::stat`, that is, current controller state and FK ([ref](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/c836b944889889329687499c9839d0d3b1dbd471/libraries/YarpPlugins/CartesianControlServer/RateThreadImpl.cpp)). Should we wrap this output in a timestamp so that downstreams (client device) can query the instant at which FK was computed? Better yet: just reuse the joint timestamp streamed by ControlBoardWrapper by implementing a chained call to the [`yarp::dev::IPreciselyTimed`](http://www.yarp.it/classyarp_1_1dev_1_1IPreciselyTimed.html) interface. In this manner, we receive FK data along with the instant at which joint data acquisition was performed.

PeterBowman, 2019-04-16:
The [`yarp::dev::ICartesianControl` interface](http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html) provides a `yarp::os::Stamp` output parameter in several methods, e.g. `getPose(yarp::sig::Vector & x, yarp::sig::Vector & o, yarp::os::Stamp * stamp = NULL)`:

> stamp: the stamp of the encoders employed to compute the pose.

PeterBowman, 2019-04-16:
Out of these three client-server communication ways:

- RPC (1-to-1 with acknowledgment)
- command (1-to-N)
- streaming (N-from-1)

we can attach timestamps to streaming messages (joint device -> cartesian device -> downstreams) and RPC messages (cartesian device -> downstreams). The former propagates encoder acquisition times, the latter - just the instant in which a response has been prepared by the server and prior to being sent back to the RPC client. I began the streaming part implementation at ae622f79a8d4b991659b689c78d7956ff27f0302.

However, I believe we are not interested in the RPC component, we don't care whether the request has been processed sooner or later. ASWJ, logging state is our main concern, therefore just the `stat()` method can take advantage from acquisition timestamps among all commands available in the `ICartesianControl` interface. Moreover, `getPose()` is precisely the only method in YARP's cartesian control client/server architecture that uses stamps, and even more - these are passed on as a function parameter.

Proposal: repurpose current `stat()` command

```c++
ICartesianControl::stat(int * state, std::vector<double> & x);
```

as

```c++
ICartesianControl::stat(std::vector<double> & x, int * state = NULL, double * timestamp = NULL);
```

Thanks to defaulting pointer parameters, we don't need to adhere to the usual verbose boilerplate:

```c++
int state; // I don't want to add this if I'm never gonna use it!
std::vector<double> x;
iCartesianControl->stat(&state, x);
```

Usually, we'd be happy with `iCartesianControl->stat(x)`, ignoring current CC state. @jgvictores what do you think?

jgvictores, 2019-04-16:
> timestamps to streaming messages (joint device -> cartesian device -> downstreams) 

Great!

> ICartesianControl::stat(std::vector<double> & x, int * state = NULL, double * timestamp = NULL);

Let's go for it, no problem!

PeterBowman, 2019-04-16:
> Proposal: repurpose current `stat()` command

My bad, it's `ICartesianControl::stat(int & state, std::vector<double> & x);`, notice the pass-by-reference on `state`. I'd rather pass a value pointer, though, as explained in my earlier comment (references are rarely used with primitive types).

closed 2019-04-22 by jgvictores