roboticslab-uc3m/kinematics-dynamics #177: Develop kdl-from-csv.py

jgvictores, 2019-04-29:
### Background
At https://github.com/roboticslab-uc3m/teo-configuration-files/blob/23703e1e29182a43c491efe04d4f5a9427d6008a/share/kinematics/leftArmKinematics.ini#L3-L4 we can see that certain transformations (namely, `H0`) were computed via `asibot-main/example/cpp/testTinyMath.cpp`. If fact, this has to become [exampleYarpTinyMath.cpp](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/530561b45c238a6f18cffa30d20d90bb25ac587c/examples/cpp/exampleYarpTinyMath/exampleYarpTinyMath.cpp). It was quite a manual process.

### Proposal
Similar to `roboview-from-csv.py` at https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/530561b45c238a6f18cffa30d20d90bb25ac587c/scripts/python, develop `kdl-from-csv.py`, which can directly parse files off https://github.com/roboticslab-uc3m/teo-developer-manual/tree/a83379613c4d8ccd6183375fd6916412521c205e/csv (specifically, the `dh-root-*.csv` files) to generate transformations (namely, `H0` and `HN`).

jgvictores, 2019-04-30:
Done. Merged at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/cd5a144fec8a9c088b0923a4a39572db387c3b23

rsantos88, 2019-04-30:
Wow!! Amazing!! 
I only replaced the `/home/yo/repos` by `/home/teo/repos` (more common according to the route we usually use)

jgvictores, 2019-04-30:
> I only replaced the `/home/yo/repos` by `/home/teo/repos` (more common according to the route we usually use)

Hot-fix applied at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/ebd008765a6ab49e6d87f50dab0a715f9de8decd

closed 2019-04-30 by jgvictores