roboticslab-uc3m/kinematics-dynamics #179: Reaching joint position limits affects TCP orientation

PeterBowman, 2019-05-30:
Steps to reproduce:

1. Launch *streamingDeviceController*.
1. Move the robot TCP so that a joint limit is hit (motion stops, cannot proceed further).
1. Move away from that joint limit.

 We observed that the last step may result in a different TCP orientation even if that was not intended. All subsequent commands do not seem to take this into account and the TCP behaves erratically, even moving in the opposite direction.

Follows up https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/173.

PeterBowman, 2024-03-29:
Perhaps adding a look-ahead would help, as in [tiago_spnav_teleop](https://github.com/asrob-uc3m/tiago_spnav_teleop/blob/ffe23cd10a6840beecf4ad0e41846a3bdbaf547c/src/tiago_spnav_teleop.cpp#L157). That is, calculate the new step the robot would process and move to, but then calculate another one to decide whether the first should be ignored (e.g. we are approaching joint limits or a singularity) or not.

jgvictores, 2024-03-30:
Definitely could be considered dated (but imho, first replicating and then advancing can be a good approach) but this of reminds me of "S. Kim, A. Shukla and A. Billard, Catching Objects in Flight, in IEEE Transactions on Robotics, vol. 30, no. 5, pp. 1049-1065, Oct. 2014", https://doi.org/10.1109/TRO.2014.2316022 -> I recall one of the constraints that was difficult to attain was complying with robot joint velocity limits (looking through the paper, they also mention singularities and joint limits).
- https://www.youtube.com/watch?v=H0ds54G53kI
- https://www.youtube.com/watch?v=M413lLWvrbI

jgvictores, 2024-03-30:
PS: From https://scholar.google.com/scholar?cites=5957588769371584553&as_sdt=2005&sciodt=0,5&hl=es "trajectory tracking control" seems like important keywords.