roboticslab-uc3m/kinematics-dynamics #184: [BasicCartesianControl] Not finding remote ports with OpenraveYarpPaintSquares

RaulFdzbis, 2020-01-11:
Getting this error when trying to run BasicCartesianControl on a computer with Bionic and Openrave 0.9:

```bash
[info] DeviceDriverImpl.cpp:20 open(): Subdevice BasicCartesianControl
[debug] DeviceDriverImpl.cpp:11 open(): BasicCartesianControl config: (H0 (0 1 0 0 0 0 1 -0.3469 1 0 0 0.4982 0 0 0 1)) (HN (0 0 -1 -0.0975 -1 0 0 0 0 1 0 0 0 0 0 1)) (angleRepr axisAngle) (device BasicCartesianControl) (from "/usr/local/share/teo-configuration-files/contexts/kinematics/fixedTrunk-rightArm-fetch.ini") (ik st) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_1 (offset -90.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_2 (offset -90.0) (D -0.32901) (A 0.0) (alpha -90.0) (mass 1.750625) (cog 0 0 0) (inertia 0 0 0)) (link_3 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_4 (offset 0.0) (D -0.215) (A 0.0) (alpha -90.0) (mass 2.396) (cog 0 0 0) (inertia 0 0 0)) (link_5 (offset -90.0) (D 0.0) (A -0.09) (alpha 0.0) (mass 0.3) (cog 0 0 0) (inertia 0 0 0)) (local "/BasicCartesianControl/teoSim/rightArm") (name "/teoSim/rightArm/CartesianControl") (numLinks 6) (remote "/teoSim/rightArm") (single_threaded 1) (subdevice BasicCartesianControl).
yarp: Port /BasicCartesianControl/teoSim/rightArm/rpc:o active at tcp://192.168.1.57:10009/
yarp: Port /BasicCartesianControl/teoSim/rightArm/command:o active at tcp://192.168.1.57:10010/
yarp: Port /BasicCartesianControl/teoSim/rightArm/stateExt:i active at tcp://192.168.1.57:10011/
[ERROR]Problem connecting to /teoSim/rightArm/rpc:i, is the remote device available?
[ERROR]Problem connecting to /teoSim/rightArm/command:i, is the remote device available?
[ERROR]Problem connecting to /teoSim/rightArm/stateExt:o, is the remote device available?
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
[error] DeviceDriverImpl.cpp:55 open(): robot device not valid: remote_controlboard.
yarpdev: ***ERROR*** driver <BasicCartesianControl> was found but could not open
[error] DeviceDriverImpl.cpp:38 open(): cannot make <BasicCartesianControl>
[error] DeviceDriverImpl.cpp:49 open(): cartesianControlDevice not valid
yarpdev: ***ERROR*** driver <CartesianControlServer> was found but could not open
[ERROR]yarpdev: ***ERROR*** device not available.
[INFO]Suggestions:
[INFO]+ Do "yarpdev --list" to see list of supported devices.
[INFO]+ Or append "--verbose" option to get more information.
```

If i am not wrong the problem comes from not finding the ports `/teoSim/rightArm/*`. After running OpenraveYarpPaintSquares the available ports are:

``` bash
~$ yarp name list
registration name /OpenraveYarpPluginLoader/rpc:s ip 192.168.1.57 port 10002 type tcp
registration name /OpenraveYarpPluginLoader/state:o ip 192.168.1.57 port 10003 type tcp
registration name /openraveYarpPaintSquares/rpc:s ip 192.168.1.57 port 10008 type tcp
registration name /root ip 192.168.1.57 port 10000 type tcp
registration name /teoSim/rightLacqueyFetch/internalFinger/command:i ip 192.168.1.57 port 10005 type tcp
registration name /teoSim/rightLacqueyFetch/internalFinger/rpc:i ip 192.168.1.57 port 10004 type tcp
registration name /teoSim/rightLacqueyFetch/internalFinger/state:o ip 192.168.1.57 port 10006 type tcp
registration name /teoSim/rightLacqueyFetch/internalFinger/stateExt:o ip 192.168.1.57 port 10007 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
```

Are `/teoSim/rightLacqueyFetch/*` equivalent to `/teoSim/rightArm/*` ?. Should we change the name in the configuration files? Or am I doing something wrong?

The output of OpenraveYarpPlugins

jgvictores, 2020-01-11:
> Are `/teoSim/rightLacqueyFetch/*` equivalent to `/teoSim/rightArm/*` ?. Should we change the name in the configuration files? Or am I doing something wrong?

No, most probably `teoSim` died. Could you please restart your computer and copy the full output of launching `teoSim` ?

RaulFdzbis, 2020-01-13:
Here you go:

```bash
~$ teoSim
2020-01-13 10:44:55,460 openrave [WARN] [plugindatabase.h:645 InterfaceBasePtr OpenRAVE::RaveDatabase::Create] Failed to create name viewerrecorder, interface module
[info] OpenraveYarpPluginLoader.cpp:22 OpenraveYarpPluginLoader(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:25 OpenraveYarpPluginLoader(): Found yarp network.
yarp: Port /OpenraveYarpPluginLoader/rpc:s active at tcp://10.118.40.231:10002/
yarp: Port /OpenraveYarpPluginLoader/state:o active at tcp://10.118.40.231:10003/
[debug] OpenraveYarpPluginLoader.cpp:85 main(): [env openrave/teo_lacqueyFetch.robot.xml open --device controlboardwrapper2 --subdevice YarpOpenraveControlboard --robotIndex 0 --allManipulators]
[debug] OpenraveYarpPluginLoader.cpp:123 main(): env: 'openrave/teo_lacqueyFetch.robot.xml'
2020-01-13 10:44:55,498 openrave [WARN] [libopenrave.cpp:799 __cxx11::string OpenRAVE::RaveGlobal::FindLocalFile] could not find file "openrave/teo_lacqueyFetch.robot.xml"
2020-01-13 10:44:55,498 openrave [WARN] [environment-core.h:492 Load] load failed on file openrave/teo_lacqueyFetch.robot.xml
[debug] OpenraveYarpPluginLoader.cpp:133 main(): Could not load 'openrave/teo_lacqueyFetch.robot.xml' environment, attempting via yarp::os::ResourceFinder.
2020-01-13 10:44:56,235 openrave [INFO] [ForceSensor.cpp:59 ForceSensor] Creating a Force Sensor
2020-01-13 10:44:56,235 openrave [INFO] [ForceSensor.cpp:59 ForceSensor] Creating a Force Sensor
[success] OpenraveYarpPluginLoader.cpp:143 main(): Loaded '/usr/local/share/teo-openrave-models/openrave/teo_lacqueyFetch.robot.xml' environment.
[debug] OpenraveYarpPluginLoader.cpp:150 main(): open[0]: [open --device controlboardwrapper2 --subdevice YarpOpenraveControlboard --robotIndex 0 --allManipulators]
[info] OpenraveYarpPluginLoader.cpp:168 Open(): Checking for yarp network...
[success] OpenraveYarpPluginLoader.cpp:175 Open(): Found yarp network.
[debug] OpenraveYarpPluginLoader.cpp:182 Open(): config: (allManipulators) (device controlboardwrapper2) (robotIndex 0) (subdevice YarpOpenraveControlboard)
[info] OpenraveYarpPluginLoader.cpp:185 Open(): penv: 0x557c8a5b2780
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:27 open(): config: (allManipulators) (device YarpOpenraveControlboard) (manipulatorIndex 0) (name "/teoSim/rightLacqueyFetch/thumb") (penv {128 39 91 138 124 85 0 0 160 42 91 138 124 85 0 0}) (robotIndex 0) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:64 open(): Get JointPtr for manipulatorIDs[0]: 34 (rightLacqueyFetch/ThumbProximal)
[debug] DeviceDriverImpl.cpp:64 open(): Get JointPtr for manipulatorIDs[1]: 35 (rightLacqueyFetch/ThumbDistal)
[debug] DeviceDriverImpl.cpp:80 open(): pcontrol: 0x557c8c0b6f50, IdealController
[info] DeviceDriverImpl.cpp:86 open(): Detected idealcontroller, switch to genericmulticontroller.
[debug] DeviceDriverImpl.cpp:99 open(): pcontrol: 0x557c8a808190, genericmulticontroller
[debug] DeviceDriverImpl.cpp:116 open(): Attach individual controller for manipulatorIDs[0]: 34
[debug] DeviceDriverImpl.cpp:116 open(): Attach individual controller for manipulatorIDs[1]: 35
[INFO]created device <YarpOpenraveControlboard>. See C++ class roboticslab::YarpOpenraveControlboard for documentation.
[info] IPositionControlImpl.cpp:15 getAxes(): Reporting 2 axes are present
[DEBUG]joints parameter is 2
[info] IPositionControlImpl.cpp:15 getAxes(): Reporting 2 axes are present
[info] IPositionControlImpl.cpp:15 getAxes(): Reporting 2 axes are present
[INFO]/teoSim/rightLacqueyFetch/thumb : no ROS initialization required 
[INFO]/teoSim/rightLacqueyFetch/thumb  initting YARP initialization 
yarp: Port /teoSim/rightLacqueyFetch/thumb/rpc:i active at tcp://10.118.40.231:10004/
yarp: Port /teoSim/rightLacqueyFetch/thumb/command:i active at tcp://10.118.40.231:10005/
yarp: Port /teoSim/rightLacqueyFetch/thumb/state:o active at tcp://10.118.40.231:10006/
yarp: Port /teoSim/rightLacqueyFetch/thumb/stateExt:o active at tcp://10.118.40.231:10007/
[INFO]created wrapper <controlboardwrapper2>. See C++ class ControlBoardWrapper for documentation.
[success] OpenraveYarpPluginLoader.cpp:358 Open(): Valid yarp plugin (id 0).
[DEBUG]ControlBoardWrapper2: 'period' parameter missing, using default thread period = 20ms 
[DEBUG]opening controlBoardWrapper2 subdevice
[debug] DeviceDriverImpl.cpp:27 open(): config: (allManipulators) (device YarpOpenraveControlboard) (manipulatorIndex 1) (name "/teoSim/rightLacqueyFetch/externalFinger") (penv {128 39 91 138 124 85 0 0 160 42 91 138 124 85 0 0}) (robotIndex 0) (subdevice YarpOpenraveControlboard)
[debug] DeviceDriverImpl.cpp:64 open(): Get JointPtr for manipulatorIDs[0]: 36 (rightLacqueyFetch/ExternalFingerProximal)
[debug] DeviceDriverImpl.cpp:64 open(): Get JointPtr for manipulatorIDs[1]: 37 (rightLacqueyFetch/ExternalFingerDistal)
[debug] DeviceDriverImpl.cpp:80 open(): pcontrol: 0x557c8a808190, genericmulticontroller
[info] DeviceDriverImpl.cpp:92 open(): Detected genericmulticontroller, which will be used.
[debug] DeviceDriverImpl.cpp:99 open(): pcontrol: 0x557c8a808190, genericmulticontroller
[debug] DeviceDriverImpl.cpp:116 open(): Attach individual controller for manipulatorIDs[0]: 36
[debug] DeviceDriverImpl.cpp:116 open(): Attach individual controller for manipulatorIDs[1]: 37
[... cut ...]
[success] OpenraveYarpPluginLoader.cpp:159 main(): Open ids: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 
```

Everythink looks ok I think. Maybe it's a problem of `OpenraveYarpPaintSquares`?. The command Im using to launch `BasicCartesianControl` is:

```bash
yarpdev --device BasicCartesianControl --name /teoSim/rightArm/CartesianControl --from /usr/local/share/teo-configuration-files/contexts/kinematics/fixedTrunk-rightArm-fetch.ini --local /BasicCartesianControl/teoSim/rightArm --remote /teoSim/rightArm --ik st --angleRepr axisAngle
```

jgvictores, 2020-01-13:
Okay, this codebase is quite old. [openraveYarpPaintSquares.py](https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/01a85a9205189f235a40027d0eca73aa20b611e2/examples/python/openraveYarpPaintSquares.py) in fact requires you NOT to use `teoSim`.

Recommendations:
- Via the `x` button of the `openrave` viewer, please close any instance of `teoSim` and/or `openrave` (the use of the GUI rather than `ctrl+C` is due to https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/59).
- Type `yarp clean`
Then just `openraveYarpPaintSquares.py` and then `BasicCartesianControl`.

RaulFdzbis, 2020-01-13:
Restarted my computer, used `yarp clean` just in case, and tried `openraveYarpPaintSquares.py` and still no working. Same errors and same ports available.

RaulFdzbis, 2020-01-13:
As always was a problem of having multiple residual installations of the same package, thanks to @jgvictores  for helping me to solve the problem.

RaulFdzbis, 2020-01-13:
Still, `BasicCartesianControl` is not working. Now, when i try to run it, it gives me a segmentation fault error. This is the output of `BasicCartesianControl`:

```bash
~$ yarpdev --device BasicCartesianControl --name /teoSim/rightArm/CartesianControl --from /usr/local/share/teo-configuration-files/contexts/kinematics/fixedTrunk-rightArm-fetch.ini --local /BasicCartesianControl/teoSim/rightArm --remote /teoSim/rightArm --ik st --angleRepr axisAngle
[info] DeviceDriverImpl.cpp:20 open(): Subdevice BasicCartesianControl
[debug] DeviceDriverImpl.cpp:11 open(): BasicCartesianControl config: (H0 (0 1 0 0 0 0 1 -0.3469 1 0 0 0.4982 0 0 0 1)) (HN (0 0 -1 -0.0975 -1 0 0 0 0 1 0 0 0 0 0 1)) (angleRepr axisAngle) (device BasicCartesianControl) (from "/usr/local/share/teo-configuration-files/contexts/kinematics/fixedTrunk-rightArm-fetch.ini") (ik st) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_1 (offset -90.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_2 (offset -90.0) (D -0.32901) (A 0.0) (alpha -90.0) (mass 1.750625) (cog 0 0 0) (inertia 0 0 0)) (link_3 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_4 (offset 0.0) (D -0.215) (A 0.0) (alpha -90.0) (mass 2.396) (cog 0 0 0) (inertia 0 0 0)) (link_5 (offset -90.0) (D 0.0) (A -0.09) (alpha 0.0) (mass 0.3) (cog 0 0 0) (inertia 0 0 0)) (local "/BasicCartesianControl/teoSim/rightArm") (name "/teoSim/rightArm/CartesianControl") (numLinks 6) (remote "/teoSim/rightArm") (single_threaded 1) (subdevice BasicCartesianControl).
yarp: Port /BasicCartesianControl/teoSim/rightArm/rpc:o active at tcp://10.118.40.231:10009/
yarp: Port /BasicCartesianControl/teoSim/rightArm/command:o active at tcp://10.118.40.231:10010/
yarp: Port /BasicCartesianControl/teoSim/rightArm/stateExt:i active at tcp://10.118.40.231:10011/
yarp: Sending output from /BasicCartesianControl/teoSim/rightArm/rpc:o to /teoSim/rightArm/rpc:i using tcp
yarp: Sending output from /BasicCartesianControl/teoSim/rightArm/command:o to /teoSim/rightArm/command:i using udp
yarp: Receiving input from /teoSim/rightArm/stateExt:o to /BasicCartesianControl/teoSim/rightArm/stateExt:i using udp
[INFO]created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[info] DeviceDriverImpl.cpp:107 open(): numRobotJoints: 6.
[info] DeviceDriverImpl.cpp:149 open(): Joint 0 limits: [-98.100000,106.000000] [-30.000000,30.000000]
[info] DeviceDriverImpl.cpp:149 open(): Joint 1 limits: [-75.500000,22.400000] [-30.000000,30.000000]
[info] DeviceDriverImpl.cpp:149 open(): Joint 2 limits: [-80.100000,57.000000] [-30.000000,30.000000]
[info] DeviceDriverImpl.cpp:149 open(): Joint 3 limits: [-99.600000,98.400000] [-30.000000,30.000000]
[info] DeviceDriverImpl.cpp:149 open(): Joint 4 limits: [-80.400000,99.600000] [-30.000000,30.000000]
[info] DeviceDriverImpl.cpp:149 open(): Joint 5 limits: [-115.100000,44.700000] [-30.000000,30.000000]
[debug] DeviceDriverImpl.cpp:135 open(): config: (H0 (0 1 0 0 0 0 1 -0.3469 1 0 0 0.4982 0 0 0 1)) (HN (0 0 -1 -0.0975 -1 0 0 0 0 1 0 0 0 0 0 1)) (angleRepr axisAngle) (device KdlSolver) (from "/usr/local/share/teo-configuration-files/contexts/kinematics/fixedTrunk-rightArm-fetch.ini") (ik st) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_1 (offset -90.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_2 (offset -90.0) (D -0.32901) (A 0.0) (alpha -90.0) (mass 1.750625) (cog 0 0 0) (inertia 0 0 0)) (link_3 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_4 (offset 0.0) (D -0.215) (A 0.0) (alpha -90.0) (mass 2.396) (cog 0 0 0) (inertia 0 0 0)) (link_5 (offset -90.0) (D 0.0) (A -0.09) (alpha 0.0) (mass 0.3) (cog 0 0 0) (inertia 0 0 0)) (local "/BasicCartesianControl/teoSim/rightArm") (maxs (106.0 22.4 57.0 98.4 99.6 44.7)) (mins (-98.1 -75.5 -80.1 -99.6 -80.4 -115.1)) (name "/teoSim/rightArm/CartesianControl") (numLinks 6) (remote "/teoSim/rightArm") (single_threaded 1) (subdevice BasicCartesianControl).
[info] DeviceDriverImpl.cpp:139 open(): kinematics: none.ini [none.ini]
||| did not find none.ini
yarp: cannot read from 
[debug] DeviceDriverImpl.cpp:150 open(): fullConfig: (H0 (0 1 0 0 0 0 1 -0.3469 1 0 0 0.4982 0 0 0 1)) (HN (0 0 -1 -0.0975 -1 0 0 0 0 1 0 0 0 0 0 1)) (angleRepr axisAngle) (device KdlSolver) (from "/usr/local/share/teo-configuration-files/contexts/kinematics/fixedTrunk-rightArm-fetch.ini") (ik st) (link_0 (offset 0.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_1 (offset -90.0) (D 0.0) (A 0.0) (alpha -90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_2 (offset -90.0) (D -0.32901) (A 0.0) (alpha -90.0) (mass 1.750625) (cog 0 0 0) (inertia 0 0 0)) (link_3 (offset 0.0) (D 0.0) (A 0.0) (alpha 90.0) (mass 0) (cog 0 0 0) (inertia 0 0 0)) (link_4 (offset 0.0) (D -0.215) (A 0.0) (alpha -90.0) (mass 2.396) (cog 0 0 0) (inertia 0 0 0)) (link_5 (offset -90.0) (D 0.0) (A -0.09) (alpha 0.0) (mass 0.3) (cog 0 0 0) (inertia 0 0 0)) (local "/BasicCartesianControl/teoSim/rightArm") (maxs (106.0 22.4 57.0 98.4 99.6 44.7)) (mins (-98.1 -75.5 -80.1 -99.6 -80.4 -115.1)) (name "/teoSim/rightArm/CartesianControl") (numLinks 6) (remote "/teoSim/rightArm") (single_threaded 1) (subdevice BasicCartesianControl).
[info] DeviceDriverImpl.cpp:154 open(): numLinks: 6 [1]
[... cut ...]
[INFO]created device <KdlSolver>. See C++ class roboticslab::KdlSolver for documentation.
[info] DeviceDriverImpl.cpp:175 open(): numSolverJoints: 6.
[INFO]created device <BasicCartesianControl>. See C++ class roboticslab::BasicCartesianControl for documentation.
yarp: Port /teoSim/rightArm/CartesianControl/rpc:s active at tcp://10.118.40.231:10016/
yarp: Port /teoSim/rightArm/CartesianControl/command:i active at tcp://10.118.40.231:10017/
yarp: Port /teoSim/rightArm/CartesianControl/state:o active at tcp://10.118.40.231:10018/
Segmentation fault (core dumped)
```

jgvictores, 2020-01-13:
1. I would have the intuition it's related to the 6 vs 8 (not sure if joints or links) discrepancy.
2. Among several runs, I did observe a `[error] ICartesianControlImpl.cpp:35 stat(): getEncoders failed.`

@PeterBowman Any ideas?

closed 2020-01-13 by PeterBowman

jgvictores, 2020-01-14:
@PeterBowman thanks!

RaulFdzbis, 2020-01-14:
I have just test it and is working like a charm, thanks!

commit 73c2f481 (2020-01-13): Account for missing transform options to avoid dereferencing nullptr

Regression caused by d6ca7c1 and 3f0b861. Fixes #184.