roboticslab-uc3m/kinematics-dynamics #187: Python bindings

RaulFdzbis, 2020-06-23:
I don't know why, but looks like the python bindings are not being installed in my computer, am i doing something wrong? 

Following the official [installation guide](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/develop/doc/kinematics-dynamics-install.md) the outputs I get are the followings:

```bash
~…/kinematics-dynamics/bindings (develop)$ cmake .. -DCREATE_PYTHON=ON
-- Found YCM: /usr/local/share/cmake/YCM (found version "0.11.20200107.6-20200107.6+git597f2b1")
CMake Deprecation Warning at /usr/local/lib/cmake/YARP/YARPConfig.cmake:178 (message):
  The 'OS' component is deprecated.  Use 'os' instead, and replace
  YARP::YARP_OS with YARP::YARP_os.
Call Stack (most recent call first):
  CMakeLists.txt:41 (find_package)

-- Found YARP: /usr/local/lib/cmake/YARP (found version "3.3.0+12-20191211.4+git5b635d878")
-- Setting up installation of roboticslab-kinematics-dynamics.ini to /usr/local/share/yarp/config/path.d folder.
--  [ ] plugin AmorCartesianControl (dependencies unsatisfied: "AMOR_API_FOUND") (ENABLE_AmorCartesianControl)
--  [x] plugin AsibotSolver (ENABLE_AsibotSolver)
--  [x] plugin BasicCartesianControl (ENABLE_BasicCartesianControl)
--  [x] plugin CartesianControlClient (ENABLE_CartesianControlClient)
--  [x] plugin CartesianControlServer (ENABLE_CartesianControlServer)
--  [x] plugin KdlSolver (ENABLE_KdlSolver)
-- ROBOTICSLAB_YARP_DEVICES found, building streamingDeviceController with sensor support
-- Configuring done
-- Generating done
-- Build files have been written to: /home/raul/repos/kinematics-dynamics/bindings

```

and for make install:

```bash
~…/kinematics-dynamics/bindings (develop)$ sudo make install; sudo ldconfig; cd # install and go home
[  2%] Built target KdlVectorConverterLib
[  4%] Built target KinematicRepresentationLib
[ 16%] Built target ScrewTheoryLib
[ 18%] Built target TrajectoryLib
[ 26%] Built target AsibotSolver
[ 33%] Built target BasicCartesianControl
[ 39%] Built target CartesianControlClient
[ 46%] Built target CartesianControlServer
[ 54%] Built target KdlSolver
[ 56%] Built target YarpTinyMathLib
[ 61%] Built target haarDetectionController
[ 66%] Built target keyboardController
[ 75%] Built target streamingDeviceController
[ 79%] Built target transCoords
[ 81%] Built target gtest
[ 83%] Built target gtest_main
[ 86%] Built target testScrewTheory
[ 88%] Built target testKinRepresentation
[ 90%] Built target testKdlSolver
[ 93%] Built target testBasicCartesianControl
[ 95%] Built target testKdlSolverFromFile
[ 97%] Built target testAsibotSolverFromFile
[100%] Built target testKdlTrajectory
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/share/yarp/config/path.d/roboticslab-kinematics-dynamics.ini
-- Up-to-date: /usr/local/share/roboticslab-kinematics-dynamics/plugins/roboticslab-kinematics-dynamics.ini
-- Up-to-date: /usr/local/lib/cmake/ROBOTICSLAB_KINEMATICS_DYNAMICS/ROBOTICSLAB_KINEMATICS_DYNAMICSConfigVersion.cmake
-- Up-to-date: /usr/local/lib/cmake/ROBOTICSLAB_KINEMATICS_DYNAMICS/ROBOTICSLAB_KINEMATICS_DYNAMICSConfig.cmake
-- Up-to-date: /usr/local/lib/cmake/ROBOTICSLAB_KINEMATICS_DYNAMICS/ROBOTICSLAB_KINEMATICS_DYNAMICSTargets.cmake
-- Installing: /usr/local/lib/cmake/ROBOTICSLAB_KINEMATICS_DYNAMICS/ROBOTICSLAB_KINEMATICS_DYNAMICSTargets-release.cmake
-- Up-to-date: /usr/local/lib/libKdlVectorConverterLib.so
-- Up-to-date: /usr/local/include/KdlVectorConverter.hpp
-- Up-to-date: /usr/local/lib/libKinematicRepresentationLib.so
-- Up-to-date: /usr/local/include/KinematicRepresentation.hpp
-- Up-to-date: /usr/local/lib/libScrewTheoryLib.so
-- Up-to-date: /usr/local/include/MatrixExponential.hpp
-- Up-to-date: /usr/local/include/ProductOfExponentials.hpp
-- Up-to-date: /usr/local/include/ScrewTheoryIkProblem.hpp
-- Up-to-date: /usr/local/include/ConfigurationSelector.hpp
-- Up-to-date: /usr/local/lib/libTrajectoryLib.so
-- Up-to-date: /usr/local/include/ICartesianTrajectory.hpp
-- Up-to-date: /usr/local/include/ITrajectory.hpp
-- Up-to-date: /usr/local/include/KdlTrajectory.hpp
-- Up-to-date: /usr/local/include/ICartesianControl.h
-- Up-to-date: /usr/local/include/ICartesianSolver.h
-- Up-to-date: /usr/local/lib/roboticslab-kinematics-dynamics/AsibotSolver.so
-- Installing: /usr/local/share/roboticslab-kinematics-dynamics/plugins/AsibotSolver.ini
-- Up-to-date: /usr/local/lib/roboticslab-kinematics-dynamics/BasicCartesianControl.so
-- Installing: /usr/local/share/roboticslab-kinematics-dynamics/plugins/BasicCartesianControl.ini
-- Up-to-date: /usr/local/lib/roboticslab-kinematics-dynamics/CartesianControlClient.so
-- Installing: /usr/local/share/roboticslab-kinematics-dynamics/plugins/CartesianControlClient.ini
-- Up-to-date: /usr/local/lib/roboticslab-kinematics-dynamics/CartesianControlServer.so
-- Installing: /usr/local/share/roboticslab-kinematics-dynamics/plugins/CartesianControlServer.ini
-- Up-to-date: /usr/local/lib/roboticslab-kinematics-dynamics/KdlSolver.so
-- Installing: /usr/local/share/roboticslab-kinematics-dynamics/plugins/KdlSolver.ini
-- Up-to-date: /usr/local/lib/libYarpTinyMathLib.so
-- Up-to-date: /usr/local/include/YarpTinyMath.hpp
-- Up-to-date: /usr/local/bin/haarDetectionController
-- Up-to-date: /usr/local/bin/keyboardController
-- Up-to-date: /usr/local/bin/streamingDeviceController
-- Up-to-date: /usr/local/bin/transCoords
-- Up-to-date: /usr/local/lib/libgtest.a
-- Up-to-date: /usr/local/lib/libgtest_main.a
-- Up-to-date: /usr/local/include/gtest
-- Up-to-date: /usr/local/include/gtest/gtest-death-test.h
-- Up-to-date: /usr/local/include/gtest/gtest.h
-- Up-to-date: /usr/local/include/gtest/internal
-- Up-to-date: /usr/local/include/gtest/internal/gtest-tuple.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-param-util-generated.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-type-util.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-filepath.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-string.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-death-test-internal.h
-- Up-to-date: /usr/local/include/gtest/internal/custom
-- Up-to-date: /usr/local/include/gtest/internal/custom/gtest.h
-- Up-to-date: /usr/local/include/gtest/internal/custom/gtest-port.h
-- Up-to-date: /usr/local/include/gtest/internal/custom/gtest-printers.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-param-util-generated.h.pump
-- Up-to-date: /usr/local/include/gtest/internal/gtest-param-util.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-port.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-port-arch.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-linked_ptr.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-internal.h
-- Up-to-date: /usr/local/include/gtest/internal/gtest-type-util.h.pump
-- Up-to-date: /usr/local/include/gtest/internal/gtest-tuple.h.pump
-- Up-to-date: /usr/local/include/gtest/gtest_pred_impl.h
-- Up-to-date: /usr/local/include/gtest/gtest-param-test.h
-- Up-to-date: /usr/local/include/gtest/gtest-param-test.h.pump
-- Up-to-date: /usr/local/include/gtest/gtest-printers.h
-- Up-to-date: /usr/local/include/gtest/gtest-spi.h
-- Up-to-date: /usr/local/include/gtest/gtest_prod.h
-- Up-to-date: /usr/local/include/gtest/gtest-message.h
-- Up-to-date: /usr/local/include/gtest/gtest-test-part.h
-- Up-to-date: /usr/local/include/gtest/gtest-typed-test.h
-- Up-to-date: /usr/local/share/roboticslab-kinematics-dynamics/ymanager.ini
-- Up-to-date: /usr/local/share/roboticslab-kinematics-dynamics/contexts/streamingDeviceController/streamingDeviceController.ini
-- Up-to-date: /usr/local/share/roboticslab-kinematics-dynamics/applications/streamingDeviceController/streamingSharedControl.xml
-- Up-to-date: /usr/local/share/roboticslab-kinematics-dynamics/contexts/testAsibotSolverFromFile/testAsibotSolverFromFile.ini
-- Up-to-date: /usr/local/share/roboticslab-kinematics-dynamics/contexts/testKdlSolverFromFile/testKdlSolverFromFile.ini
~$ 
```

I tried using the python module of kinematics-dynamics but it just give me the not found error.

PeterBowman, 2020-06-23:
> ```shell
> ~…/kinematics-dynamics/bindings (develop)$ cmake .. -DCREATE_PYTHON=ON
> ```

Here, you are building the main C++ project, just as you would with the usual `build/` directory structure from the *kinematics-dynamics: Installation from Source Code* step. You need to enter the `bindings/` directory, **create a `build/` directory inside**, and issue the `cmake ..` command from within. You must have missed the `mkdir build && cd build` steps in the install guide.

jgvictores, 2020-06-23:
@PeterBowman +1

It's as documented at https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/master/doc/kinematics-dynamics-install.md#install-bindings ([perma](https://github.com/roboticslab-uc3m/kinematics-dynamics/blob/01b6609e79dc16f8f882f104669a2c58a49897f0/doc/kinematics-dynamics-install.md#install-python-bindings))

RaulFdzbis, 2020-06-24:
Yep! That was it now it works, thank you very much!

The problem was due to that mkdir command not having the -p flag, and having already created the build folder some time in the past. When executing  `mkdir build && cd build` the mkdir crashed and  `cd build` was never performed. Maybe is a good idea to add the -p flag to the install instructions.

PeterBowman, 2020-06-24:
cd's are so old-school, what if:

```sh
src=~/repos/kinematics-dynamics/bindings
build=$src/build
mkdir -p $build
cmake -H $src -B $build -DCREATE_PYTHON=ON
make -C $build -j$(nproc)  # compile
sudo make -C $build install
sudo ldconfig
```

RaulFdzbis, 2020-06-24:
hahahaha cd's are still cool.

That looks good, it take me a moment to understand what everything was, but at the end it's pretty straightforward, if you think is a better option im in :D

closed 2020-07-14 by PeterBowman