roboticslab-uc3m/kinematics-dynamics #188: ImportError: dynamic module does not define init function

RaulFdzbis, 2020-06-24:
Getting the following error while trying to import kinematics-dynamics:

```bash
~$ python
Python 2.7.17 (default, Apr 15 2020, 17:20:14) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import kinematics_dynamics
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/kinematics_dynamics.py", line 17, in <module>
    _kinematics_dynamics = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/kinematics_dynamics.py", line 16, in swig_import_helper
    return importlib.import_module('_kinematics_dynamics')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: dynamic module does not define init function (init_kinematics_dynamics)
```

After i quick internet search i have tried with python3, but it looks like it is only installed in the 2.7 version. Any ideas why this may be happening?

PeterBowman, 2020-06-24:
I can't reproduce this, same Python version. A quite similar error was reported at https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/147#issuecomment-368910149, but that one was clearly showing a Python 2.7 vs 3.4 mismatch.

Can you paste here the output of `grep ^PYTHON ~/repos/kinematics-dynamics/bindings/build/CMakeCache.txt | grep -v INTERNAL`?

Also, what is your SWIG version (`swig -version`)?

RaulFdzbis, 2020-06-25:
Here you go!

```bash
~$ grep ^PYTHON ~/repos/kinematics-dynamics/bindings/build/CMakeCache.txt | grep -v INTERNAL
PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3.6
PYTHON_INCLUDE_DIR:PATH=/usr/include/python3.6m
PYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
PYTHON_LIBRARY_DEBUG:FILEPATH=PYTHON_LIBRARY_DEBUG-NOTFOUND
~$ swig -version

SWIG Version 1.3.29

Compiled with g++ [x86_64-unknown-linux-gnu]
Please see http://www.swig.org for reporting bugs and further information

```

Maybe should I reinstall changing python filepath to 2.7?

PeterBowman, 2020-06-25:
Thanks!

> ```shell
> SWIG Version 1.3.29
> ```

That's a really ancient version of SWIG, probably not compatible with YARP anymore. I'm pretty sure you have SWIG 3.x installed, so please try again with `swig3.0 -version`. If not, do `sudo apt install swig3.0` and proceed again. Also, take a look at the advanced configuration of kin-dyn bindings via `ccmake` (press `t`) and check your `SWIG_xxx` variables. Mine look like these:

```
SWIG_DIR=/usr/share/swig3.0
```
SWIG_DIR=/usr/share/swig3.0

RaulFdzbis, 2020-06-25:
This is my ccmake output of bindings:

```bash
 PYTHON_EXECUTABLE                /usr/bin/python3.6            
 PYTHON_INCLUDE_DIR               /usr/include/python3.6m     
 PYTHON_LIBRARY                   /usr/lib/x86_64-linux-gnu/libpython3.6m.so
 PYTHON_LIBRARY_DEBUG             PYTHON_LIBRARY_DEBUG-NOTFOUND
....
 SWIG_DIR                         /usr/share/swig3.0
 SWIG_EXECUTABLE                  /usr/bin/swig3.0
 SWIG_VERSION                     3.0.12
```
I didnt find the ``CMAKE_INSTALL_PYTHONDIR`` variable. In my case Kinematics-dynamics was installed on ``/usr/local/lib/python3/dist-packages/``, what I did was:

```
sudo ln -s /usr/local/lib/python3/dist-packages/_kinematics_dynamics.so /usr/local/lib/python2.7/dist-packages/
sudo ln -s /usr/local/lib/python3/dist-packages/kinematics_dynamics.py /usr/local/lib/python2.7/dist-packages/
```

PeterBowman, 2020-06-25:
I don't think you should mix Python 2.x and 3.x dist-packages as it would probably cause compat issues, I'd remove those symlinks you added. So, what is the actual problem here? You built kin-dyn bindings for Python 3, which got properly installed, now you should be able to load the corresponding module via `python3`. It obviously won't work that way if you launch the Python 2 interpreter instead. Did you want to compile Python 2 bindings? In that case, please tweak CMake variables to point at Python 2.7 directories and executables and you should be fine.

RaulFdzbis, 2020-06-25:
I tried to do an import using python3.6 and it works:

```bash
~…/bindings/build (develop)$ python3.6
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import kinematics_dynamics
>>> 
```

So maybe reinstall changing paths to 2.7?

PeterBowman, 2020-06-25:
> So maybe reinstall changing paths to 2.7?

👍

RaulFdzbis, 2020-06-25:
Thought the sym links would work haha. I will try it and close the issue if everything goes right.

Thank you very much Bartek!

RaulFdzbis, 2020-06-25:
Deleted symlinks, changed ccmake bindings to

```
 PYTHON_EXECUTABLE                /usr/bin/python2.7            
 PYTHON_INCLUDE_DIR               /usr/include/python2.7      
 PYTHON_LIBRARY                   /usr/lib/x86_64-linux-gnu/libpython2.7.so
```
and is working. Thank you again @PeterBowman

closed 2020-06-25 by RaulFdzbis