roboticslab-uc3m/kinematics-dynamics #192: Orientation system to encode and decode pose of pitch-roll

rsantos88, 2021-06-09:
I am trying to solve point 7 of the issue https://github.com/HUMASoft/yarp-devices/issues/7 but I am in doubt whether `kinematicRepresentationLib` has the possibility to work with the pitch-roll representation form. Previously, our project has worked with `POLAR_AZIMUTH`  as a orientation system to encode and decode the pose of inclination and orientation. However, if we wanted to put positive and negative angles in pitch and roll, what would be the correct orientation system that would respect the signs and angles?

PeterBowman, 2021-06-09:
What is exactly pitch-roll, i.e. which angles get affected (XYZ, local or global) and what is the sequence? Is it actually different from the current polar-azimuthal system? What happens if you pass negative values? Does it need normalization if the azimuthal angle exceed the [-180, 180] range?

rsantos88, 2021-06-10:
Currently we pass the roll-pitch values connecting to the port `../rpc_transform:s` using the `movj(roll pitch)` command. 
The device is launched by adding the parameters  `--coordRepr none --angleRepr polarAzimuth --angularUnits degrees`.
Therefore, internally movj works with a vector describing a three-dimensional pose in coordinate_system :: CARTESIAN and orientation_system :: AXIS_ANGLE_SCALED systems. When we want to retrieve the roll-pitch values indicated by RPC through this port, we need to apply [decodePose](https://github.com/HUMASoft/yarp-devices/blob/sensor3DMGX510/libraries/YarpPlugins/SoftNeckControl/ICartesianControlImpl.cpp#L112) currently configured with `orientation_system::POLAR_AZIMUTH`.
The problem is that the values are distorted, thus leaving the result of the `targetPose` vector: 
`movj 0 0` -> targetPose = [0, -0]
`movj 10 0` -> targetPose = [10 -0]
`movj 0 10` -> targetPose = [0 -0]
`movj 10 10` -> targetPose = [10 10]
`movj -10 0` -> targetPose = [10 180]
`movj 0 -10` -> targetPose = [0 0 ]
`movj -10 -10` -> targetPose = [10 170]

It is surely an absurd error, but I am trying to follow the structure with which it was commanded using `movj` with inclination and orientation. I think that since these values are always positive, the `targetPose` correctly returned its values when passing through the decodePose function with Polar-Azimuth orientation system.

rsantos88, 2021-06-13:
Thanks for the explanation. It makes perfect sense that the function tries to transform the output into always positive angles, preserving the equivalence in terms of polar-azimuth orientation. The point is that in the case that we work, both the sensor and the controller work with roll-pitch, also the target that we want to introduce. Therefore, I would like the decodePose function to respect exactly the same values entered on its output.
Testing your proposal to use EULER_ZYZ, I've  verified that I would still get unwanted transforms for what I am looking for. 
For example, the cases that wouldn't match (as I would like it to be) are:

```bash
input> 0.000000 -10.000000 0.000000
encodePose> 0.000000 0.000000 0.000000 0.000000 -0.174533 0.000000
decodePose> -180.000000 10.000000 180.000000

input> 10.000000 -10.000000 0.000000
encodePose> 0.000000 0.000000 0.000000 0.015231 -0.174089 0.174089
decodePose> -170.000000 10.000000 -180.000000

input> -10.000000 -10.000000 0.000000
encodePose> 0.000000 0.000000 0.000000 -0.015231 -0.174089 -0.174089
decodePose> 170.000000 10.000000 180.000000 
```
I think that on second thought, perhaps the most effective solution is to continue introducing inclination-orientation values that match perfectly with the polar-azimuth input with which we have been working until now and make the conversion to roll-pitch before passing it to the controller. . In this way, our system would continue to use the same way to represent the orientation as in previous controls.

rsantos88, 2021-06-16:
> I think that on second thought, perhaps the most effective solution is to continue introducing inclination-orientation values that match perfectly with the polar-azimuth input with which we have [...]

Finally, this will be the solution to be applied. I close this issue.

closed 2021-06-16 by rsantos88