roboticslab-uc3m/kinematics-dynamics #195: Impedance control in cartesian space

PeterBowman, 2023-02-01:
As a follow-up to https://github.com/roboticslab-uc3m/kinematics-dynamics/issues/194, let's add some spring-like behavior to the [*ftCompensation* app](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/f5010f0e4ee430ff50424d31bceb83521e446bd8/programs/ftCompensation) now that it knows how to convert FT sensor data into cartesian force commands while performing gravity compensation.

PeterBowman, 2023-02-02:
Done at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/488028bcb8d9043c3cd6f208de4eded5e857e4ef. There is a lot of friction mainly in the elbow joint that should be somehow compensated in the low-level (https://github.com/roboticslab-uc3m/yarp-devices/issues/261), but for now it seems to work well in orientations (Y and Z axes) with this configuration:

```
ftCompensation --from ftCompensation-stump.ini --mode wrench \
               --sensorName ch0 --sensorRemote /jr3 \
               --cartesianRemote /teo/leftArm/CartesianControl \
               --linGain 1 --rotGain 1 \
               --linStiffness 250 --rotStiffness 10 \
               --linDamping 0 --rotDamping 1
```

Note this could have been implemented as an add-on for the GCMP command in BasicCartesianControl. Such attempt was initiated in https://github.com/roboticslab-uc3m/kinematics-dynamics/compare/6e78a53...b8ed4ca and explained in https://github.com/roboticslab-uc3m/yarp-devices/issues/258, but discarded (for now) to focus on low-level joint impedance control.

closed 2023-02-02 by PeterBowman

jgvictores, 2023-02-06:
Just for the historic, and completely low priority, would a video be possible (maybe of this vs gcmp)?