roboticslab-uc3m/kinematics-dynamics #198: Rename `ICartesianControl::movi` to `pose`?

PeterBowman, 2024-05-18:
There used to be a streaming `pose` command: https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/7af8fc70a0bf14aa37bdc04e492068a49264b678 (https://github.com/roboticslab-uc3m/kinematics-dynamics/pull/117). Before its deprecation at https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/469d1035f717f1cad5ba819268f25adf8efb0f19, the documentation stated:

```cxx
/**
 * @brief Achieve pose (velocity control)
 *
 * Move to desired position, computing the error with respect to the current pose. Then,
 * perform numerical differentiation and obtain the final velocity increment (as in @ref twist).
 *
 * @param x 6-element vector describing desired instantaneous pose in cartesian space;
 * first three elements denote translation (meters), last three denote rotation (radians).
 * @param interval Time interval between successive command invocations, expressed in seconds
 * and used for numerical differentiation with desired/current poses.
 */
void pose(const std::vector<double> & x, double interval);
```

The `movi` command was introduced later (https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/87a9c94b11a41e0cc860963fb2c67c58a564455b) with the following description:

```cxx
/**
 * @brief Achieve pose
 *
 * Move to desired position instantaneously, no further intermediate calculations are
 * expected other than computing the inverse kinematics.
 *
 * @param x 6-element vector describing desired instantaneous pose in cartesian space;
 * first three elements denote translation (meters), last three denote rotation (radians).
 */
void movi(const std::vector<double> & x);
```

I find the `movi` name a bit difficult to understand at first glance. Also, ROS users may question the naming of this command due to the existence of Pose, Twist and Wrench geometry message types ([ref](https://docs.ros2.org/latest/api/geometry_msgs/index-msg.html)). Since the information conveyed is in fact a pose, then why not use this name instead?

cc @jgvictores @mercerebo1

jgvictores, 2024-05-19:
LGTM!

BTW, maybe we should be more explicit on the representation of orientation (compact axis/angle, id I recall correctly)?

closed 2024-05-23 by PeterBowman

PeterBowman, 2025-04-16:
@roboticslab-uc3m/teo the `movi` command is being removed now, please rename it to `pose` in your code.